sengsational/cwhelper #7: Slot equals bug

sengsational, 2023-12-27:
The slot equals is broken.

This line isn't really even required, but if it's there, it needs to be checking to make sure both variables are NOT null:
https://github.com/sengsational/cwhelper/blob/06401820156d6424c24319f322e957dd3d3a3828/CwHelperC/src/org/cwepg/hr/Slot.java#L302

So should read "            if (end != null && other.end != null){"

And then this line, the comparison is wrong.  It should be "greater than" (if the time is greater than a minute, then it's not equal):
https://github.com/sengsational/cwhelper/blob/06401820156d6424c24319f322e957dd3d3a3828/CwHelperC/src/org/cwepg/hr/Slot.java#L306

So should read "                if (differenceMs > 60000){"

And likewise, the same two changes need to be made here:
https://github.com/sengsational/cwhelper/blob/06401820156d6424c24319f322e957dd3d3a3828/CwHelperC/src/org/cwepg/hr/Slot.java#L319

and here:
https://github.com/sengsational/cwhelper/blob/06401820156d6424c24319f322e957dd3d3a3828/CwHelperC/src/org/cwepg/hr/Slot.java#L323

TPeterson94070, 2023-12-27:
Understood.  I'll make these edits and test.

TPeterson94070, 2023-12-27:
They worked!

sengsational, 2024-01-03:
I made if off the sandbar.  Good news that you've got these changes in.  Thanks for getting those things in and testing them.  If you push them and do a pull request, I'll merge to the feature branch.  Then, if we're satisfied this feature is done, I'll merge to the master branch.

TPeterson94070, 2024-01-03:
I've done the pr. Those updates are included, but I'm confused about some of the earlier ones that I thought had been incorporated before. I assume that any repeats will be effectively ignored. 

https://github.com/sengsational/cwhelper/issues/7#issuecomment-1874785099

sengsational, 2024-01-04:
Resolved.  It was broken on 10/26/2010 in an attempt to make small differences in end-event timestamps irrelevant.  It was only broken for 13 years.  I suspect I was testing "dump compare" to see if it was working, and neglected to reverse the changes.  The lesson here is to put a "TODO" marker in the code, so these kinds of things won't get left in the code.
[image]

sengsational, 2024-01-04:
Resolved.

closed 2024-01-04 by sengsational