sengsational/cwhelper #14: Either broaden RF.PID definitions for non-US locale or disable RF.PID for them.

TPeterson94070, 2024-02-07:
The present code assumes us-bcast or us-cable locale for the RF.PID parameters.  Since the "channel numbers" are not relevant outside of ATSC locales, the preferred route, IMO, is to disable RF.PID if not an ATSC locale.

sengsational, 2024-02-08:
That sounds like a good thing since, if I'm understanding this right, outside of ATSC locales, RF.PID does not make sense.

Is this another case where we filter the channel data when we build our lineup?  But it doesn't mean we skip the channels, it just means the channels get alternative data populated?  Maybe the channel object will need to have an instance variable or a method to let us know it's of this non-ATSC type?

TPeterson94070, 2024-02-28:
Currently, CW_EPG only permits RF.PID mode on us-bcast stations.  It's the only mode possible for the old tuners and can be enabled for new units if they have us-bcast OTA stations.  All non us-bcast stations in the S.D. data appear to be either cable or DTV-B, both of which CW_EPG handles the same as VC-only us-bcast.  AFAIK, this is a tenable approach so this issue can be closed.

closed 2024-02-28 by TPeterson94070