sengsational/cwhelper #15: No need for second (secure) TinyServer

TPeterson94070, 2024-02-09:
Since it turned out that there's no royal road to setting up an https server on the private LAN, I've moved on to looking at stashing the csv files for CW_EPG_Remote in the user's Google Drive folder instead of fetching them from the app data folder.  That means that the second server that you set up, intended to become SSL protected, isn't needed.

Can we just drop it?

sengsational, 2024-02-09:
Sure.  We can always dig it out of source control if we need it.

TPeterson94070, 2024-02-12:
Do you want to make the snip?  If not, please point me to the place.

sengsational, 2024-02-13:
I can do this one.

sengsational, 2024-02-13:
This is the commit where this was added [31cc4be](https://github.com/sengsational/cwhelper/commit/31cc4beeddacb953f690b2fa7910a4645e25adfd).  It's probably ok to just remove the server startup and leave the file in the code base.

TPeterson94070, 2024-02-13:
Okay, thanks, after you've done that I will pull down a refresh from the Repository and generate a new CWHelper so that I can unleash the latest and greatest on our beta test guys.

TPeterson94070, 2024-02-13:
Great!  Thanks, I'll gin up a new beta test package for the gang.

closed 2024-02-28 by sengsational