sengsational/cwhelper #16: No longer need for "optional" html files

TPeterson94070, 2024-02-20:
We can clean up the stdout log by removing the check for (and use of) the 3 patch files that were the way Terry added/edited the right-click CWHelper menu actions:
```
Tue Feb 13 10:06:13 PST 2024 Could not find optional html file [C:\ProgramData\CW_EPG\\CWHelpervcr.html].
Tue Feb 13 10:06:13 PST 2024 Could not find optional html file [C:\ProgramData\CW_EPG\\CWHelpersettings.html].
Tue Feb 13 10:06:13 PST 2024 Could not find optional html file [C:\ProgramData\CW_EPG\\CWHelpershutdown.html].
```
OK?  If so, do you want to do this?  Or point to the spot(s) for action?

sengsational, 2024-02-24:
I created a separate branch and created a pull request rather than just commit it.  I'll merge later if this looks good.

sengsational, 2024-02-24:
Actually, this wasn't what I intended to do from a source control perspective.  I used feature_s as the base, so it contains all of the changes made there.  I should have switched to master then made the new branch.  Where's the forehead smack emoji?

sengsational, 2024-02-24:
OK, Here's a pull request against master.  I'm just doing this to get used to making GitHub do what it can do.  In this case, pushing this to master as if it were a more urgent bug fix rather than put it in the feature branch with all of the things that might need validation. https://github.com/sengsational/cwhelper/pull/18

sengsational, 2024-02-26:
Merged the change into main.

closed 2024-02-26 by sengsational