sengsational/cwhelper #21: Feature request: Add metadata to transport stream

sengsational, 2024-03-14:
@TPeterson94070 I need some help with the mapping between the fields in `CW_EPG_RECORDS` and the fields that are used in garybuhrmaster's implementation.

TPeterson94070, 2024-03-14:
I just emailed my suggestions, repeated here:
Gtb   name | Cw_record.mdb   field
'Category' | SHOWTYPE
'ChannelAffiliate' | n/a
'ChannelImageURL' | n/a
'ChannelName' | CHANNEL   (should use PSIP name, but it’s not in History)
'ChannelNumber' | VIR+’.’+SUB
'EndTime' | ORIG_END
'EpisodeNumber' | n/a   (actually is in SUBTITLE, but…)
'EpisodeTitle' | SUBTITLE
'FirstAiring' | NEW
'ImageURL' | n/a
'OriginalAirdate' | AIRDATE
'ProgramID' | EPISODE
'RecordEndTime' | ENDTIME
'RecordStartTime' | STARTTIME
'RecordSuccess' | CONFIRMED
'Resume' | n/a
'SeriesID' | n/a   (not sure what this is supposed to be)
'StartTime' | ORG_START
'Synopsis' | DESCRIPTION
'Title' | TITLE

sengsational, 2024-03-14:
Except for "NEW", I found all of the fields in my CW_EPG_RECORDS table.

TPeterson94070, 2024-03-14:
Oops.  Sorry "NEW" is in the EPG dataset and not supplied in History.  I guess we can just skip it.

TPeterson94070, 2024-03-15:
On further thought, I realized that CWHelper needs to skip 'RecordSuccess' also, because CONFIRMED only gets set "True" when CW_EPG runs after the capture has been done so it will always be "False" when CWHelper is writing the file.

closed 2024-03-21 by sengsational