sengsational/cwhelper #23: Original air date sometimes shows 1969

sengsational, 2024-03-24:
CWHelper puts a “0” for OriginalAirdate in movie metadata by default when the database AIRDATE field isn't parseable.  That’s undesirable because it translates into the end of 1969 when made human-readable.  

We could instead use January 1 of the MOVIE_YEAR field for movies (if AIRDATE isn't popluated).  If neither AIRDATE or MOVIE_YEAR are parseable, do not include the OriginalAirdate in the metadata.

TPeterson94070, 2024-03-24:
Looks good!

sengsational, 2024-03-24:
I made the changes and pushed to master on https://github.com/sengsational/cwhelper/commit/542ebe96f4846b4ce4e2ea02a8e73373eee4ec1a

The version number is inaccurate in the one I committed, but in the Google Drive files area, it's right (+1 from the last one).

closed 2024-03-24 by sengsational

TPeterson94070, 2024-03-24:
I think you got this one.  😊