sengsational/cwhelper #26: Metadata not copied if file name contains ampersand

sengsational, 2024-03-27:
The database has %26 for ampersands so direct query `where filename = 'Something & Something` does not find the record.

TPeterson94070, 2024-03-27:
For the use case you mention, only the %26 would apply since CWHelper is already translating the %20 back to <space> and %2A would never occur.

TPeterson94070, 2024-03-27:
BTW, this is done to prevent an issue with CWHelper's parsing of the /save command.  Ampersands are perfectly OK in filenames, so the %26 could be decoded prior to executing the /save command.

sengsational, 2024-03-28:
I version of CW_EPG 5.4.149 and later, file names will contain "&" as opposed to "%26", so no change is required in CwHelper.

closed 2024-03-28 by sengsational

TPeterson94070, 2024-03-28:
Funny that this was Issue #26. 😉  And funny that CWHelper was doing the right thing all along and CW_EPG was just putting a bogus filename in its History.

sengsational, 2024-03-31:
I noticed to swirly confluence of "26" too (cue twilight zone music).