sengsational/cwhelper #31: SeriesID metadata not inclulded if database lacks SUBTITLE entry

sengsational, 2024-04-04:
`SeriesID` JSON is populated from database field `SUBTITLE`, but if that does not exist, `SeriesID` does not get populated in the metadata.

I proposed this metadata should be generated as "C00" appended to a hash of the TITLE field.

TPeterson94070, 2024-04-04:
It should be `C00EN` prefixed on the hash of the  `TITLE` field.

sengsational, 2024-04-05:
There's a bit more processing now.  When it gets to work on `Title`, if the `SeriesID` isn't populated, it puts in the default plus hash of the TITLE db field.  I didn't test it, so I HOPE it will do that.

https://github.com/sengsational/cwhelper/commit/2904b79765e751618b4e17da5b7c0fb3a1fc3a46

TPeterson94070, 2024-04-05:
Success.  Congrats!!
[image]

closed 2024-11-08 by sengsational