sengsational/cwhelper #42: Build process requires IDE runnable jar process

sengsational, 2024-11-17:
The build process requires the IDE "export runnable jar" process instead of just making the runnable jar.

sengsational, 2024-11-17:
Updated BuildExeFromClassFiles.java so that it now creates the runnable jar. See https://github.com/sengsational/cwhelper/commit/49019d7e1dd495cc8f43c0460a92cbd8ced06998

sengsational, 2024-11-17:
@TPeterson94070 I left your specifics in the class file, but they're commented out.  So when you pull this into your environment, you'll have to uncomment those, and comment mine.  We probably should have a local (non-source control) config file, but this will work for now.

sengsational, 2024-11-18:
The implementation had a bug where backslashes were used when creating zip entries.  That's why java -jar ConstructedJarFile.jar said "Could not find or load main class", even though the main class was there.  Made the change here: https://github.com/sengsational/cwhelper/commit/e7d8522abc9745da0d59c5e0c2cc061911192f4c

With this update to JarUpdater.java, it should allow us to run the builder program one time and get the exe, the zipped exe, and the jar.

TPeterson94070, 2024-11-19:
Roger that.  It works now as long as I tell the 2024-09 Eclipse IDE to limit Java to JRE 1.8.