sengsational/cwhelper #46: Event logging minute can be different from target event minute

sengsational, 2024-11-22:
Events are scheduled on the even minute (no seconds).  When the system realizes an event is ready to occur, the actual event time value is posted to the event object.  This actual event time can be a few seconds before the scheduled event time. 

When logging for the occurs, the actual even time is extracted.  If the actual event time is in the preceding minute, the database gets an incorrect value because the seconds are truncated.

One solution would be to change the event logging to database to include seconds.  Another solution would be to add a few seconds to bring the event into the correct minute, but keep the existing truncation.

sengsational, 2024-11-22:
I took a look at the existing code and found that the `updateCaptureEndEvent()` timestamp already has 15 seconds added to it, with the comment that it truncates seconds:

https://github.com/sengsational/cwhelper/blob/69f8f0089cec95ca3901f5a433be3651edbaf51b/CwHelperC/src/org/cwepg/hr/CaptureDetails.java#L125

The `insertCaptureStartEvent()` has no such adjustment, which is probably why the one minute discrepancy was noted on a start event.

sengsational, 2024-11-22:
Checked in a change that puts seconds into the database for actual start and actual end times.  Also removed the 15 second addition noted in the comment above, because truncation is no longer an issue.

https://github.com/sengsational/cwhelper/commit/bd717a11b7cda899af0f1dfadaa5eb5ea6540415

A quick test shows the seconds made it into the database.
[image]

TPeterson94070, 2024-12-01:
This is working.

closed 2024-12-01 by TPeterson94070