sengsational/cwhelper #47: Metadata not applied when the file name changes

sengsational, 2024-11-22:
In some situations, the file name gets changed on the fly.  The change is where "_00n" is added, where "n" is an integer.

When the process that adds metadata runs, it looks in the database for the original file name (without the "_00n"), and does not find it, so metadata is not added.

One solution would be to expand the database lookup so that it finds both the original file name, as well as the renamed versions.

sengsational, 2024-11-23:
Expanded the SQL query for metadata addition and added data range to query.  Checked into master.

https://github.com/sengsational/cwhelper/commit/55821bf33b2a467b55fc97500cb6f33c30c62a77

TPeterson94070, 2024-12-01:
You got it, thanks!

closed 2024-12-01 by TPeterson94070