sengsational/cwhelper #48: Replacement capture process leaves zero byte files

sengsational, 2024-11-29:
Before http capture recordings begin, they check the availability of the tuner.  If it's busy, an attempt is made to find a replacement tuner, which is fine, but the unsuccessful recapture should not leave a zero byte file, and it should not require that the replacement capture increment the file name with "001", etc.

Also, because the original (failed) recording does not create a record in the captures table, there's no reason to attempt to update that record.  At this point, just logs a warning, but no need to attempt that update at all.

TPeterson94070, 2024-11-29:
I think you've got it all.

sengsational, 2024-11-29:
I added a call to a new method that attempts to delete the zero byte file. https://github.com/sengsational/cwhelper/commit/45c04fb829b461ae0000fb27f95086c9ead48b84

I didn't do anything with the end event.  There's a bunch activities that probably are not needed, but don't hurt anything.  And there's a bunch of things that happen in the end capture processing to make sure things stay tidy.  Most are probably not required, and if we want to tone-down the logging, we can do that, but I'd rather just let it plow through all of the tidy-up processes.

TPeterson94070, 2024-12-01:
OK with me.

closed 2024-12-01 by TPeterson94070