sengsational/cwhelper #50: hdhomerun_config.exe save command output includes unhandled error messages

sengsational, 2024-12-19:
Normally the 'save' command outputs a string of dots, each representing some time when the save command worked without issue.  If transient recording issues occur, non-dot symbols appear, and the program appropriately just counts those symbols and reports that.  But the save command can respond with messages including the word "error", where it makes no sense to report the non-dot symbols.

A solution would be to check for the word "error" in the output from the save command.  If found, place a log entry that's easy to spot.  If not found use existing functionality.

sengsational, 2024-12-19:
Fix checked into master https://github.com/sengsational/cwhelper/commit/45c04fb829b461ae0000fb27f95086c9ead48b84

sengsational, 2024-12-21:
In cases where the exe returns the string of dots and later runs out of space, the error message should not include the dots.

sengsational, 2024-12-21:
Changed the message to log just the error message (presuming that's the last thing in the report)

https://github.com/sengsational/cwhelper/commit/d54cd984f6d592a456e6064c949c2608886b8744

TPeterson94070, 2025-01-10:
This is working.

closed 2025-01-10 by TPeterson94070