sengsational/cwhelper #52: Discover command from hdhomerun.com can return hdhomerun units on inaccessable IP addresses

sengsational, 2024-12-29:
If the local hdhomerun_config.exe discover command is not available, the program goes out to http://ipv4-api.hdhomerun.com/discover which returns the local IP addresses and devices.  If the household has multiple subnets, not all tuners will necessarily be accessible.

One solution would be to include only devices on the subnet of the machine running CwHelper.

sengsational, 2024-12-30:
I moved the discover of the machine's IP out in a separate class, "NetworkDetails", out of TunerHdhr, where it was before.  Then I was able to call it from TunerManager (which is the thing that creates the tuners).

https://github.com/sengsational/cwhelper/commit/d93c1402303c24a672b6ab22f6524728c3a29cd8

TPeterson94070, 2025-01-10:
I will confirm this fix via Steve's log after he's installed this version.

sengsational, 2025-02-03:
Has Steve seen the bug since he's updated?

TPeterson94070, 2025-02-03:
I’m pretty sure that it’s fixed, but I’ll check with him.  I’ve been letting it slide because he’s been preoccupied with his wife’s recent health issue.

TPeterson94070, 2025-02-05:
The fix works.

closed 2025-02-05 by TPeterson94070