sengsational/cwhelper #54: Machine can sleep before the next capture can configure the device

sengsational, 2025-01-01:
If a machine is configured to sleep immediately, it's possible that the machine can sleep between two successive trips through the main loop.

The first trip through the loop ends the recording.  The next trip through the loop starts the next recording.  But the next recording is not "active" at the time the first recording ends.  When the first recording ends, it therefore allows sleep since there's no active recordings present.

One solution would be to bypass allow sleep if there was a pending recording (this along with the existing reasons to bypass).

sengsational, 2025-01-01:
Checked into master: https://github.com/sengsational/cwhelper/commit/8f1aa25f7274d93155139e249d4df70516234b5e

TPeterson94070, 2025-01-01:
N.B.:  All machines are likely configured effectively to "sleep immediately" because the sleep timer is not reset during a capture. So, unless the timeout is unreasonably (IMO) long it will have expired before the capture finishes.

sengsational, 2025-01-03:
As a recap:
We had the nodding-off issue with machines that sleep immediately because there was a problem with the response from "get stream info".  That problem caused a delay that was too long, and the machine went to sleep.

So there was a change made to the code such that if there was a pending activity, the "allow sleep" would not be processed.

But then we got to the root cause of the problem of the "get stream info": quick succession of setting the tuner then asking the tuner for stream info doesn't work....there must be a delay.  So there was a change made to the code to put a short (one second) delay between setting the channel and getting stream info: https://github.com/sengsational/cwhelper/commit/35833f78b5aee597bef6ec5a20745838058ec249

TPeterson94070, 2025-01-10:
I concur.  It's an okay belt and suspenders thing.

closed 2025-01-10 by TPeterson94070