sengsational/cwhelper #56: Splash screen closes too quickly when started from jar file

sengsational, 2025-01-10:
Right now, there's a thread that starts at the very beginning on program initiation, waits 2 seconds (configurable), then closes the splash screen.  

Another alternative would be to shut off the splash screen when the program's tray icon is added.

TPeterson94070, 2025-01-10:
The test version works well from both jar and exe.  However, we need to call the splash closer method even if the tray icon switch is set to "false".

sengsational, 2025-01-11:
Updated to include closing the splash screen if the tray icon config is false.

https://github.com/sengsational/cwhelper/commit/b00072dec025a409353180a56678355c75fc0543

TPeterson94070, 2025-01-11:
This is working.

closed 2025-01-11 by TPeterson94070

sengsational, 2025-01-13:
I like the way this works now...the splash screen says up longer, indicating the program is still initializing.  Then the tray icon appears and then the splash disappears.