sengsational/cwhelper #59: Weak captures may be spawn backup recordings on non-mapped stations

sengsational, 2025-01-18:
Given a recording that exceeds the poor quality threshold and a backup recording is triggered, the candidate tuners for the backup recording include those that are not mapped.  I would have expected that unmapped stations would not be considered.

    Sat Jan 18 02:00:01 EST 2025 Channel alternates [3400026.0, 8580027.0, 8910031.0, 8911032.0, 8912026.0, 8913027.0,]
    
    Sat Jan 18 02:02:01 EST 2025 Checking errors interval: since start reportLength: 114
    Sat Jan 18 02:02:01 EST 2025 There were 25 errors resulting in since start severity 0.21929824 as compared to 0.1
    Sat Jan 18 02:02:01 EST 2025 Poor quality recording has been detected.  Attempting to start another recording in place of Capture [ChannelDigital [21.3 PBS NC 58.1 PBS NC fav:false sourceId: pid:3 freq:21 proto:8vsb virt:58.1 input:1 1013FADA-1 (us-bcast 21) 8581027.0]] [Slot: start [01/18/25 02:00:00] stop [01/18/25 02:59:59]] start/end [true/false] target [192.168.3.65:55100 f:\tv\21.3-250118-0200-Miss Scarlet on Masterpiece-(S05E01) The Rival.tp Miss Scarlet on Masterpiece E521DRS] There were 25 errors resulting in since start severity 0.21929824 as compared to 0.1
    Sat Jan 18 02:02:01 EST 2025 channel ChannelDigital [21.3 PBS NC 58.1 PBS NC fav:false sourceId: pid:3 freq:21 proto:8vsb virt:58.1 input:1 1013FADA-1 (us-bcast 21) 8581027.0]
    Sat Jan 18 02:02:01 EST 2025 altChannels size 6
    Sat Jan 18 02:02:01 EST 2025 Trying to schedule a backup capture: Capture [ChannelDigital [21.3 PBS NC 58.1 PBS NC fav:false sourceId: pid:3 freq:21 proto:8vsb virt:58.1 input:1 1010CC54-0 (us-bcast 21) 3400026.0]] [Slot: start [01/18/25 02:00:00] stop [01/18/25 02:59:59]] start/end [false/false] target []
    Sat Jan 18 02:02:01 EST 2025 INFO: Path [f:\tv] is writable. Temp file deleted.
    Sat Jan 18 02:02:01 EST 2025 Leaving existing path on the drive. f:\tv
    Sat Jan 18 02:02:01 EST 2025 CaptureManager.scheduleCapture(newCapture, true)
    writing capture Capture [ChannelDigital [21.3 PBS NC 58.1 PBS NC fav:false sourceId: pid:3 freq:21 proto:8vsb virt:58.1 input:1 1010CC54-0 (us-bcast 21) 3400026.0]] [Slot: start [01/18/25 02:00:00] stop [01/18/25 02:59:59]] start/end [false/false] target [192.168.3.65:0 f:\tv\21.3-250118-0200-Miss Scarlet on Masterpiece-(S05E01) The Rival.tp Miss Scarlet on Masterpiece E521DRS]
    Sat Jan 18 02:02:01 EST 2025 A backup capture was scheduled. Capture [ChannelDigital [21.3 PBS NC 58.1 PBS NC fav:false sourceId: pid:3 freq:21 proto:8vsb virt:58.1 input:1 1010CC54-0 (us-bcast 21) 3400026.0]] [Slot: start [01/18/25 02:00:00] stop [01/18/25 02:59:59]] start/end [false/false] target [192.168.3.65:0 f:\tv\21.3-250118-0200-Miss Scarlet on Masterpiece-(S05E01) The Rival.tp Miss Scarlet on Masterpiece E521DRS]

[stdout.txt](https://github.com/user-attachments/files/18465359/stdout.txt)

From channel_maps.txt, sorted alphabetically to get all WUNG PBS NC together:

    WUNGDT,PBS NC,21,58,1,0,888,3,27391,269744858,E521DRS,,HR(1013FADA-0),Y,,2
    WUNGDT,PBS NC,21,58,1,1,889,3,27391,269744858,E521DRS,,HR(1013FADA-1),Y,,2
    WUNGDT,PBS NC,21,58,1,1,889,3,27391,279144891,E521DRS,,HR(10A369BB-1),Y,,2
    WUNGDT,PBS NC,21,58,1,3,891,3,27391,279144891,E521DRS,,HR(10A369BB-3),Y,,2

No sign of 1010CC54-0.

TPeterson94070, 2025-01-18:
CWHelper does not normally interact with `channel_maps.txt`, as that is maintained by CW_EPG.  So CWHelper needs to poll `channel_maps.txt` for eligible tuners if we want it not to use any/all.

sengsational, 2025-02-03:
In the case where a device is unavailable, there was code that pulled the list of tuners in the order found in `channel_maps.txt`.  The code in the case where a weak recording was detected, though, it was not using `channel_maps.txt`.

I copied the code, modifying it slightly, so that `RecordMonitor` now uses the code already being used in `CaptureManager.run()` for unavailable device capture replacements.  It's a little different because there, there's no existing recording that we want to preserve.  In `RecordMonitor`, we preserve the existing recording, allowing it to keep running, but add another recording using another tuner that has been tuned to the same channel.

I was unable to test this right now, but the only time it would come into effect is for weak recordings, and I should get some of those 'naturally', so I'll keep lookout to see that the newly spawned captures from weak recordings work right, and start with the best tuners, according to the text file.

https://github.com/sengsational/cwhelper/commit/e33e7af952d864f9d21c2e947093412e27b270b2

sengsational, 2025-02-20:
Still no weak recordings.  This is because I began using Terry's excellent signal quality analyzer application along with it's ability to sort the `channel_maps.txt` by signal quality.  At some point, I might purposefully force some weak stations to validate the commit.

sengsational, 2025-02-22:
Verified that it is working:

    Sat Feb 22 09:10:11 EST 2025 Reading C:\Documents and Settings\All Users\Application Data\CW_EPG\\channel_maps.txt
    Sat Feb 22 09:10:11 EST 2025 Channel 9 tuner priority order for replacement: 1013FADA-0
    Sat Feb 22 09:10:11 EST 2025 Channel 9 tuner priority order for replacement: 1013FADA-1
    Sat Feb 22 09:10:11 EST 2025 Channel 9 tuner priority order for replacement: 10A369BB-0
    Sat Feb 22 09:10:11 EST 2025 Channel 9 tuner priority order for replacement: 10A369BB-1
    Sat Feb 22 09:10:11 EST 2025 Channel 9 tuner priority order for replacement: 10A369BB-2
    Sat Feb 22 09:10:11 EST 2025 Channel 9 tuner priority order for replacement: 10A369BB-3
    Sat Feb 22 09:10:11 EST 2025 channel_maps.txt priority used.
    Sat Feb 22 09:10:11 EST 2025 Found matching replacement tuner: 1013FADA-0
    Sat Feb 22 09:10:11 EST 2025 1013FADA-0 was available.
    Sat Feb 22 09:10:11 EST 2025 Created replacement [Capture [ChannelDigital [9.3 WTVI HD 42.1 WTVI HD fav:f....

closed 2025-02-22 by sengsational