sengsational/cwhelper #60: Hovering CwHelper capture tray icon can cause CwHelper crash

sengsational, 2025-02-08:
> I’m still noodling on the zombie capture icons that occasionally appear on my one system.  From the fact that hovering the mouse pointer over them causes CWHelper to crash when they disappear, I conclude that they’re somehow still attached to the app.  When that crash occurs there is no record in the log, it just stops.

>   This thought caused me to wonder if CWHelper can tally up and log existing threads at some other suitable times—perhaps a few seconds after it thinks it’s stopped all outstanding captures.  Does that make any sense?

It seems like we might have had some thread tally thing in the past.  I'll need to look.  Disconcerting that it's crashing the JVM without leaving any indication of what happened.

> Yeah, my guess is that Windows is killing the “rogue” task (brought to its attention by the mouse hover) and bringing CWHelper down with it.

> I finally caught another zombie and then used the tray menu to shut down CWHelper.  Then I restarted and restopped CWHelper to compare the log endings.  (Full logs attached)  Below are the two shutdown log sequences.  Aside from the order of entries, I see only one difference—in the first step—maybe significant?  Or maybe not?  What do you think?  If it’s not significant, I’m not sure where to look next.  Ideas?

> Shutdown with 1 active zombie contained this line:

    Interrupting WakeupEvent thread with command [Kill] runThread: Thread-WakeupEvent activeAutorun: false

> Shutdown with no zombie active contained this line:

    Interrupting WakeupEvent thread with command [Kill] runThread: (no runThread) activeAutorun: false

sengsational, 2025-02-08:
That difference indicates that the main run thread is gone (null) : https://github.com/sengsational/cwhelper/blob/e33e7af952d864f9d21c2e947093412e27b270b2/CwHelperC/src/org/cwepg/hr/WakeupEvent.java#L236

That's non-obvious Java syntax where if the first thing before the "?" is true, the first thing is used, if false, the thing after the ":" is used.

So the questions become, "Why is there an active `WakeupEvent.runThread`, (which is named `Thread-WakeupEvent`) in the case of the zombie?" and "Can we rectify that situation without a crash?"  Did you notice if there was a command line process (ie capturing a recording) in the Windows task list?

It looks like the thread ID's are aligned to me:

    Fri Jan 31 20:39 PST 2025 set wake machine timer for 5. native@0x154 WakeupEvent initialize for -auto -download -detail
    Sat Feb 01 10:44 PST 2025 cancelled wake machine timer. native@0x154 WakeupEvent run
    Sat Feb 01 10:45 PST 2025 set wake machine timer for 4. native@0x850 
    Sat Feb 01 10:46 PST 2025 set wake machine timer for 8. native@0x824 WakeupEvent
    Sat Feb 01 11:35 PST 2025 cancelled wake machine timer. native@0x850 
    Sun Feb 02 10:44 PST 2025 cancelled wake machine timer. native@0x824 WakeupEvent
    Sun Feb 02 10:45 PST 2025 set wake machine timer for 3. native@0x768 
    Sun Feb 02 10:45 PST 2025 set wake machine timer for 3. native@0x844 
    Sun Feb 02 10:45 PST 2025 set wake machine timer for 8. native@0x7a4 WakeupEvent
    Sun Feb 02 11:39 PST 2025 cancelled wake machine timer. native@0x768 
    Sun Feb 02 11:39 PST 2025 cancelled wake machine timer. native@0x844 
    Mon Feb 03 10:44 PST 2025 cancelled wake machine timer. native@0x7a4 WakeupEvent
    Mon Feb 03 10:45 PST 2025 set wake machine timer for 3. native@0x8a0 
    Mon Feb 03 10:45 PST 2025 set wake machine timer for 4. native@0x770 
    Mon Feb 03 10:46 PST 2025 set wake machine timer for 8. native@0x84c WakeupEvent
    Mon Feb 03 20:00 PST 2025 cancelled wake machine timer. native@0x8a0 
    Mon Feb 03 23:35 PST 2025 cancelled wake machine timer. native@0x770 
    Tue Feb 04 10:44 PST 2025 cancelled wake machine timer. native@0x84c WakeupEvent
    Tue Feb 04 10:45 PST 2025 set wake machine timer for 3. native@0x544 
    Tue Feb 04 10:45 PST 2025 set wake machine timer for 3. native@0x784 
    Tue Feb 04 10:45 PST 2025 set wake machine timer for 3. native@0x8c8 
    Tue Feb 04 10:45 PST 2025 set wake machine timer for 4. native@0x8e8 
    Tue Feb 04 10:45 PST 2025 set wake machine timer for 4. native@0x8f4 
    Tue Feb 04 10:46 PST 2025 set wake machine timer for 8. native@0x538 WakeupEvent run (Done).
    Tue Feb 04 20:00 PST 2025 cancelled wake machine timer. native@0x544 
    Tue Feb 04 20:00 PST 2025 cancelled wake machine timer. native@0x784 
    Tue Feb 04 20:00 PST 2025 cancelled wake machine timer. native@0x8c8 
    Tue Feb 04 22:00 PST 2025 cancelled wake machine timer. native@0x8f4 
    Tue Feb 04 23:35 PST 2025 cancelled wake machine timer. native@0x8e8 
    Wed Feb 05 10:44 PST 2025 cancelled wake machine timer. native@0x538 WakeupEvent run
    Wed Feb 05 10:45 PST 2025 set wake machine timer for 3. native@0x7b4 
    Wed Feb 05 10:45 PST 2025 set wake machine timer for 4. native@0x860 
    Wed Feb 05 10:45 PST 2025 set wake machine timer for 4. native@0x8a0 
    Wed Feb 05 10:46 PST 2025 set wake machine timer for 8. native@0x538 WakeupEvent run (Done).
    Wed Feb 05 21:00 PST 2025 cancelled wake machine timer. native@0x7b4 
    Wed Feb 05 22:00 PST 2025 cancelled wake machine timer. native@0x8a0 
    Wed Feb 05 23:35 PST 2025 cancelled wake machine timer. native@0x860 
    Thu Feb 06 10:44 PST 2025 cancelled wake machine timer. native@0x538 WakeupEvent run
    Thu Feb 06 10:45 PST 2025 set wake machine timer for 3. native@0x8f4 
    Thu Feb 06 10:45 PST 2025 set wake machine timer for 4. native@0x55c 
    Thu Feb 06 10:46 PST 2025 set wake machine timer for 8. native@0x768 WakeupEvent
    Thu Feb 06 10:59 PST 2025 set wake machine timer for 2. native@0x558 
    Thu Feb 06 18:00 PST 2025 cancelled wake machine timer. native@0x558 
    Thu Feb 06 21:01 PST 2025 cancelled wake machine timer. native@0x8f4 
    Thu Feb 06 23:35 PST 2025 cancelled wake machine timer. native@0x55c 
    Fri Feb 07 10:44 PST 2025 cancelled wake machine timer. native@0x768 WakeupEvent
    Fri Feb 07 10:46 PST 2025 set wake machine timer for 8. native@0x714 WakeupEvent run (Done).
    Sat Feb 08 10:05 PST 2025 cancelled wake machine timer. native@0x714 

No smoking gun at this point.

TPeterson94070, 2025-02-08:
I had looked in Task Manager on previous zombie events, hoping to see a stray task but AFAICT there are none visible there when zombies exist (and none disappear along with the zombies).  That's why I began suspecting that it was an internal CWHelper thread that had gotten stuck somehow.  Recall that this phenomenon seems to show up on only one of my systems, so if you have a test to exercise Thread-WakeupEvent, maybe it would be worthwhile to run it on that system.

sengsational, 2025-02-09:
This may have nothing to do with this problem, but there is an inconsistency in the logs related to the interrupts on the main loop in `CaptureManager`.  

Generally, events are scheduled by other threads, then `requestInterrupt()` is called to make the main loop 'realize' when the next work item is to be done.  There is a kludgy technique in that method to **manage interrupts that come in while other work is being done.**  The technique is to wait up to 10 seconds for the other work to complete. https://github.com/sengsational/cwhelper/blob/e33e7af952d864f9d21c2e947093412e27b270b2/CwHelperC/src/org/cwepg/hr/CaptureManager.java#L645-L652
If it expires all loops and still does not return from line 649, then I would expect 1) the interrupt line to NOT end with **ok.**, and 2) the message ** WARNING: CaptureManager main thread was running for more than 10 seconds after an interrupt request came in....** to be both stdout and stderr.  But there seem to be cases where only 1) is true, and we don't have the details in the log for 2), which looks "impossible" to me.

sengsational, 2025-05-15:
I'm not sure any of that made any difference to the original problem.  I think this is one where it's hard to recreate.

TPeterson94070, 2025-05-15:
I think we can close this one because the crashing only happened after there were zombie icons in the tray and the recent versions don't seem prone to creating those.  (You have the machine now that seemed vulnerable to making them and it hasn't had one for months)

closed 2025-05-15 by TPeterson94070