sengsational/cwhelper #62: Characters sent to web server can hang the program

sengsational, 2025-03-29:
Sending non UTF_8 characters can cause CwHelper to become unresponsive.

For example: "\u0043\u0068\u0027\u012f\u0301\u012f\u0301\u0064\u0069\u0069", which is unicode representation of a subtitle originally from SchedulesDirect.

sengsational, 2025-03-29:
I made a change to the input stream that heads-up the web server so that characters are converted to UTC_8.  This may cause some "?" to replace bad characters, but it might also keep the program from hanging.

    in = new BufferedReader(new InputStreamReader(new DataInputStream(client.getInputStream()), **StandardCharsets.UTF_8**));

https://github.com/sengsational/cwhelper/commit/661fb1e0810b550821368510a58fe599009503b9#diff-acc4672c489665798211dc340773c85660316e753f570d31d33fe1658c0e6596R66

TPeterson94070, 2025-03-29:
This change is compatible with the current CW_EPG and adds an extra layer of protection against passing illegal characters in filenames, e.g., via Nani's VCR.

TPeterson94070, 2025-03-30:
Please disregard my previous "oops".  This commit does work as expected with extended-Latin characters in filenames and it is necessary to the combined operation of CW_EPG and CWHelper for that.

closed 2025-03-30 by TPeterson94070