sengsational/cwhelper #63: VCR page does not encode file name

sengsational, 2025-03-31:
When the VCR web page submits captures, the file name may not be properly encoded.

sengsational, 2025-04-01:
`HtmlVcrDoc.java` and `HelpPage.xls` updated on this commit: https://github.com/sengsational/cwhelper/commit/fc2509893535b02f9447f7ba5b2a84ccedafd370

Terry sent me the files with this comment:

>This has the encodeURI added to the filename when sending the schedule request and decodeURI when fetching /captures. The former seems to work as expected. However, the /captures tables don't seem to have the encoded filenames so the decodeURI is a NOP and the non-ASCII characters look weird despite being correct in the actual files.
>  
>Eclipse insisted that there was a difference between the HelpDoc that was fetched, so I made sure to include the most-recent one in this commit.

sengsational, 2025-05-15:
I'm not sure where we stand on this one.

TPeterson94070, 2025-05-15:
Since we've removed /vcr from the right-click menu in favor of using CW_EPG's current features, this is nearly a nonissue.

closed 2025-05-15 by TPeterson94070